robopeak/rplidar_ros #12: laser_scan_matcher error with rplidar

osmancns, 2015-06-29:
I am using gmapping and it is working fine with fake odometry. I use RPlidar

But now I wanted to use laser_scan_matcher to get better odometry data. when ı used laser_scan_matcher with demo launch also it use a demo bag file , no problem it is succesfull. but when ı used to own laser bag file or used real laser data, laser_scan_matcher throw this error :

http://i.imgur.com/4SyDnIW.jpg

[ WARN] [1435132671.572301017]: Skipping scan
Strange FOV: -6.283185 rad = -360.000010 deg 
[ WARN] [1435132671.572614765]: Error in scan matching****

I am using static tf for laser to base_link.

afrancescon, 2015-08-26:
Dear osmancns,

sorry for late reply!

The problem is located at line 198 of laser_data.c (https://github.com/AndreaCensi/csm/blob/master/sm/csm/laser_data.c):

double fov = ld->max_theta - ld->min_theta;

With the GMapping assumption (zero must be centered in the scan) the resulting fov is negative. Unfortunately there are not strict convention for the scan range and any package use its own convention. 

I think the best is to add a config to RPLidar node to switch on/off the conversion needed by GMapping. Before I do that can you please check if the RPLidar node works with laser_scan_matcher  before my commit (3821d95a6e7c4979733ab24a262bb11709591f8c)?

Thanks
Alessandro

osmancns, 2015-08-26:
problem was fixed like this:

https://github.com/robopeak/rplidar_ros/pull/13

afrancescon, 2015-08-26:
Perfect... I saw this ticket still open so...

Bye
Alessandro