robopeak/rplidar_ros #15: Indigo Release

ipa-mig, 2015-08-04:
Hi @robopeak 

Do you plan on releasing this package for ROS Indigo and/or Jade?
We'd like to use this and it is just more convenient in big setups to do this.

We'd also like offer our help for releasing it to Indigo.
I'd just like to make sure that we don't get any conflicts in releasing it.

Best regards from Germany
ipa-mig.

lakehanne, 2015-08-04:
I think it works in indigo. At least I got it to work. Please see my [notes](http://lakehanne.github.io/Robo%20Peak%20Lidar/)

ipa-mig, 2015-08-04:
Sure it works. But it is not released, i.e. apt-get installable.
Having unreleased packages as a dependency is always cumbersome, as builds fail...

ipa-mig, 2015-08-21:
PUSH: Could you please provide some information how to proceed.
If I don't get any information, we'll be happy to release this package into Indigo.

We just need to stay in contact about the release procedure.

kintzhao, 2016-05-11:
thank you the organization of ipa for ever maintaining the indigo version of rplidar_ros.  we have released a new indigo version with sdk 1.5.2 which support rplidar A1 and rplidar A2 .

closed 2016-05-11 by kintzhao