robopeak/rplidar_ros #21: ROS launch error

cureinside, 2016-04-30:
I have a problem on launch the program, 
i use "roslaunch rplidar_ros view_rplidar.launch"
but it return with

> [view_rplidar.launch] is neither a launch file in package [rplidar_ros] nor is [rplidar_ros] a launch file name
> The traceback for the exception was written to the log file

I've already successfully do the catkin_make,
do i run the command on wrong directories? or what? thanks

kintzhao, 2016-05-01:
you check the package ,run:  rospack find rplidar_ros  .
if not found,  source bash in the catkin worksapce which contain the package of rplidar_ros.
such as :  sources  ~/catkin_ws/devel/setup.bash .

kintzhao, 2016-05-11:
I will close this issue, if you have any question, please comment, i will reopen it and try to solve it.

closed 2016-05-11 by kintzhao

ChristianBurth, 2018-08-22:
Hey, i ran into the same problem. 
i have sourced the bash file like suggested yet i still receive the same error.
i cant find the package rplidar_ros via rospack find either
what else could be wrong?

maxtronic, 2020-07-31:
tengo el mismo problema
[image]

yp7y-stu, 2020-12-13:
I had taken 1 day to solve this problem.
My mistake was that I didn't git under ~/catkin_ws/src.
The correct build meshod is described below.

$ mkdir ~/catkin_ws
$ mkdir ~/catkin_ws/src
$ cd ~/catkin_ws/src
$ git clone https://github.com/robopeak/rplidar_ros
$ cd ~/catkin_ws
$ ls -l /dev/ttyUSB0
$ sudo chmod 666 /dev/ttyUSB0
$ catkin_make rplidarNode
$ source ~/catkin_ws/devel/setup.bash
$ roslaunch rplidar_ros view_rplidar.launch

rishichoco, 2024-08-07:
> I had taken 1 day to solve this problem. My mistake was that I didn't git under ~/catkin_ws/src. The correct build meshod is described below.
>
> $ mkdir ~/catkin_ws $ mkdir ~/catkin_ws/src $ cd [...]

thank you bro it's worked 👌👌

sharaf-roboking, 2025-04-09:
> roslaunch rplidar_ros view_rplidar.launch

it does not work bro