robopeak/rplidar_ros #22: Port confict

polock, 2016-05-05:
It seems that turtlebot.minimal.launch uses ttyUSB0 and rplidar.launch uses it too, so I change the port parameter in the rplidar.launch to ttyUSB1 as found with the command of "ls /dev/ttyUSB*"  . But it shows that " Error, cannot bind to the specified serial port ttyUSB1". How can I solve the problem, thank you for your answer!

kintzhao, 2016-05-06:
You can follow the wiki to remap the serial port.  [How to remap the USB serial port name](https://github.com/robopeak/rplidar_ros/wiki) .  once you plugin the rplidar using USB, the ttyUSB serial port number will add one.  You can check "ls /dev/ttyUSB*" when you only repugin the rplidar.

kintzhao, 2016-05-06:
you can try to reboot your pc,  and turtlebot  add first , rplidar add next , check "ls /dev/ttyUSB*" .

polock, 2016-05-06:
Could you explain "install USB port remap: ./scripts/create_udev_rules.sh" in the wiki in detail? I got some trouble in remap.

closed 2016-05-11 by kintzhao