robopeak/rplidar_ros #25: RPLidar Scanner Mapping Issues

gautamn00, 2016-07-12:
Hello, 
I am working with the RPLidar 360-degree scanner and am having issues while trying to map in rviz. When I map, I am only seeing red dots, and no legitimate mapping. It is giving me an error message that there is no tf data and that a laser does not exist. Any help would be great. Thanks!

kintzhao, 2016-07-15:
you should paste the command which you use in terminal.  
It will be better if you give me the message of node topic tf_tree.
rostopic list 
rosnode list 
give me the tree about tf_tree.

kintzhao, 2016-07-15:
if you just launch rplidar_ros  rplidar.launch       test_rplidar.launch  view_rplidar.launch,
you can not watch the map.

gautamn00, 2016-07-15:
tf_tree is attached

rostopic list:
/camera/depth_reg
[frames.pdf](https://github.com/robopeak/rplidar_ros/files/366586/frames.pdf)
istered/camera_info
/camera/depth_registered/image_raw
/camera/rgb/camera_info
/camera/rgb/image_rect_color
/capability_server/bonds
/capability_server/events
/cmd_vel_mux/active
/cmd_vel_mux/input/navi
/cmd_vel_mux/input/safety_controller
/cmd_vel_mux/input/teleop
/cmd_vel_mux/parameter_descriptions
/cmd_vel_mux/parameter_updates
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/gateway/force_update
/gateway/gateway_info
/info
/interactions/interactive_clients
/interactions/pairing
/joint_states
/laptop_charge
/mobile_base/commands/controller_info
/mobile_base/commands/digital_output
/mobile_base/commands/external_power
/mobile_base/commands/led1
/mobile_base/commands/led2
/mobile_base/commands/motor_power
/mobile_base/commands/reset_odometry
/mobile_base/commands/sound
/mobile_base/commands/velocity
/mobile_base/controller_info
/mobile_base/debug/raw_control_command
/mobile_base/debug/raw_data_command
/mobile_base/debug/raw_data_stream
/mobile_base/events/bumper
/mobile_base/events/button
/mobile_base/events/cliff
/mobile_base/events/digital_input
/mobile_base/events/power_system
/mobile_base/events/robot_state
/mobile_base/events/wheel_drop
/mobile_base/sensors/bumper_pointcloud
/mobile_base/sensors/core
/mobile_base/sensors/dock_ir
/mobile_base/sensors/imu_data
/mobile_base/sensors/imu_data_raw
/mobile_base/version_info
/mobile_base_nodelet_manager/bond
/odom
/rosout
/rosout_agg
/scan
/scan_kinect
/tf
/tf_static
/turtlebot/incompatible_rapp_list
/turtlebot/rapp_list
/turtlebot/status
/zeroconf/lost_connections
/zeroconf/new_connections

rosnode list: 
/app_manager
/capability_server
/capability_server_nodelet_manager
/diagnostic_aggregator
/interactions
/master
/mobile_base_nodelet_manager
/record_1468532104100956954
/robot_state_publisher
/rosout
/rplidar_node_client
/turtlebot_laptop_battery
/zeroconf/zeroconf

kintzhao, 2016-07-17:
@gautamn00  I can't find the frame of laser in your tf tree. You should check your launch file .  This problem may be your configure error, please your launch file . I suggest you publish a static_transform_publisher by tf.
<node name="base2laser" pkg="tf" type="static_transform_publisher" args="0.07 0 0 0 0 0 1 /base_link /laser 50"/>

kintzhao, 2016-07-17:
rplidar_ros just for receive laser data,  not get the map.  if you use it for slam, you should add  attach it to the robot by TF at first, and run other package for slam.

kintzhao, 2016-07-17:
This is your configure error for other application, I will close it . you can paste your launch file,  if you still not find the error, i will try to  help you.

closed 2016-07-17 by kintzhao

gautamn00, 2016-07-18:
HI @kintzhao, I tried to publish a static_transform_publisher by tf and then run the program and receive this error message: [ERROR] [1468858064.611263018]: Ignoring transform for child_frame_id "laser" from authority "unknown_publisher" because of a nan value in the transform (0.000000 0.000000 0.000000) (-nan -nan -nan -nan)

gautamn00, 2016-07-18:
<launch>
  <node name="rplidarNode"          pkg="rplidar_ros"  type="rplidarNode" output="screen">
  <param name="serial_port"         type="string" value="/dev/ttyUSB0"/>  
  <param name="serial_baudrate"     type="int"    value="115200"/>
  <param name="frame_id"            type="string" value="laser"/>
  <param name="inverted"            type="bool"   value="false"/>
  <param name="angle_compensate"    type="bool"   value="true"/>
  </node>
</launch>

<launch>
  <node name="rplidarNode"          pkg="rplidar_ros"  type="rplidarNode" output="screen">
  <param name="serial_port"         type="string" value="/dev/ttyUSB0"/>  
  <param name="serial_baudrate"     type="int"    value="115200"/>
  <param name="frame_id"            type="string" value="laser"/>
  <param name="inverted"            type="bool"   value="false"/>
  <param name="angle_compensate"    type="bool"   value="true"/>
 
 </node>
  <node name="rplidarNodeClient"          pkg="rplidar_ros"  type="rplidarNodeClient" output="screen">
  </node>
</launch>

<launch>
 <node pkg="tf" type="static_transform_publisher" name="laser" args="1 1 1 x y z w  /uav/base_link_ENU /laser  100"/>
</launch>

gautamn00, 2016-07-18:
I used a new launch file and was able to see the red dots still but this time it was recognizing moving objects and surroundings. Now, I am wondering how to get a map out of it.
```
<launch>
<node pkg="tf" type="static_transform_publisher" name="base_to_laser_broadcaster" args="1 1 1 1 1 1 base_link laser 100" />
</launch>

```

kintzhao, 2016-07-19:
You should try to using hector_slam or gmapping , such some package for slam to build a map.  rplidar_ros only can measure the distance of environment.

kintzhao, 2016-07-19:
you can set a big the buffer size of scan in rviz , you will see the history laser point.

gautamn00, 2016-07-19:
Does the rp lidar publish scan data to the /scan topic? As when I try to do gmapping or hector slam I am not able to receive a map even though I record a bag file and play it.

gautamn00, 2016-07-19:
I was curious to know how to publish /LaserScan because I cannot echo it from rostopic.

gautamn00, 2016-07-19:
When I try to play back my recorded bagfile while running gmapping or hector_slam, I am unable to receive a map. It simply just says "Waiting to receive a map".

kintzhao, 2016-08-10:
At first, when you run 'launch rplidar_ros  rplidar.launch ' . you will receive the scan message topic of /scan.   when you run slam, you should configure your launch file , such as the frame_id and topic name remap,and so on.  you can check your TF tree and rqt_graph.