robopeak/rplidar_ros #26: Publishing invalid or incomplete data causes problems to the processing pipeline

naveedhd, 2016-07-15:
The node publishes scan data even when `grabScanData()` returns  `RESULT_OPERATION_FAIL` in [these lines](https://github.com/robopeak/rplidar_ros/blob/master/src/node.cpp#L260-L269).

This becomes to an issue to other nodes in processing pipeline using this scan data. One of the case is when `laserscan_to_pointcloud` will crash if scan data of `data size = 1` is published. 

As declared [here](https://github.com/ros-perception/laser_geometry/blob/indigo-devel/src/laser_geometry.cpp#L212-L215): 

`
// Extract transforms for the beginning and end of the laser scan
    ros::Time start_time = scan_in.header.stamp ;
    ros::Time end_time   = scan_in.header.stamp ;
    if(!scan_in.ranges.empty()) end_time += ros::Duration().fromSec( (scan_in.ranges.size()-1) *    scan_in.time_increment);
`

If `node_count = 1`, the `angle_increment` and `time_increment` fields of scan data will be 'inf' causing crash at above lines.

A better and safer way will be to publish ONLY if the result is successful from `grabScanData()` and further check is the size of data is 360 before publishing.

I will put up a PR for this.

naveedhd, 2016-07-15:
FYI @stonier

naveedhd, 2016-07-15:
PR #27 for this issue.

stonier, 2016-07-15:
PR #27 handles it at the ros node wrapper level because that is where @naveedhd is working, however I am wondering if other options are better (and would require input/development from the robopeak engineers):

* handle it at the sdk level
* find the root cause and if it is a bug, fix it (wherever it is - sdk/firmware)

stonier, 2016-07-15:
Just looking at the 1.5.4 commit, I wonder if [this line](https://github.com/robopeak/rplidar_ros/commit/3187e5e69722cf05514125234a2129ccfecd2b7d#diff-cc06ad9625decbfb68cc01edd08d3c1eR494) is fixing the same problem.

kintzhao, 2016-07-22:
@naveedhd  we have checked it. As  [Informational Distance Measurements]( http://www.ros.org/reps/rep-0117.html) ,publish inf message is enable.  
There is no need  Only publish 360 data.  the count according with grabScanData .
We should publish message to keep the scan publish frequent even if ascendScanData return RESULT_OPERATION_FAIL . 
 I will close the PR at first, If you have any question and comment it, i will reopen it.

bit-pirate, 2016-07-26:
@naveedhd please verify that with the commit @stonier pointed out, our problem is resolved as well. Otherwise we need to come up with a new solution.

naveedhd, 2016-07-26:
@bit-pirate @stonier the line pointed out does not address the data size. The other way round would be check data size at the subscribing nodes.

bit-pirate, 2016-07-26:
@kintzhao can a scan of size 1 be consider valid and is it a realistic use case?

If not I like to ensure the driver is not sending out such messages.

If it is, then we need to create a PR to modify that laser geometry plugin to handle scans of size 1 (start time == end time).

stonier, 2016-07-26:
+1 to bit-pirate 's comment.

@kintzhao It rather seems to me here that the driver is publishing *surprising* data that more often than not will cause problems in user's software. This requires uses to spend time debugging and then write code to correct the problem. None of which is useful time to anybody. Now, there are potentially hundreds of users and just one driver....some small corrections here could save alot of effort.

There are also other ways to inform users of things like frequency - e.g. diagnostics.

And finally, there is always the sdk underneath for a very small minority of users who for some esoteric reason, decide that they absolutely have to deal with size 1 data and the like.

bit-pirate, 2016-08-09:
*bump*

@kintzhao how can we move forward on this? This is a significant issue for us. I took us a while to hunt this down, during which we could not use our robot with the RPLIDAR reliably.

kintzhao, 2016-08-10:
@bit-pirate  "can a scan of size 1 be consider valid and is it a realistic use case?" 
What your mean: 
 the size 1 is only publish one range ,
 publish not one range and only one value is valid(>min and <max) other are inf or -inf .

I can't reproduce the case of only publish one range in /scan message. which scene and statue you can meet . when we obscure the laser around a circle, it will publish all inf message, and we consider it is fine for other application. 

can you give me  echo the message  /scan.

bit-pirate, 2016-08-10:
@kintzhao on our end @naveedhd was investigating this in depth.

@naveedhd is there data you can share with @kintzhao ? Do you know how he could reproduce the issue we had?

naveedhd, 2016-08-10:
@kintzhao **the size of range is 1** NOT one valid value. 

Previously we were printing on log when range size != 360 and from there I have this:

`[ WARN] [1461155288.335636631]: RPLidar : Node count is not equal to 360
[ WARN] [1461155288.335677183]: Node Count: 1
`

I will try to record a bag with ranges size less than 360. 

For reproducing, it also depends on the specific RPLidar being used as my assumption is older ones tend to produce more errors.

kintzhao, 2016-12-15:
But i still can not receive  your data bag.

closed 2016-12-15 by kintzhao