robopeak/rplidar_ros #28: Repo title and link to ROS wiki page

bit-pirate, 2016-07-26:
Suggesting to add this to this GitHub repo. See our fork for reference: https://github.com/yujinrobot/rplidar_ros

This help ROS users to jump back and forth between the doc (ROS wiki) and code (GitHub repo).

PS: I just created an empty page on the ROS wiki for this package: http://wiki.ros.org/rplidar_ros

kintzhao, 2016-08-15:
Thanks for your improvement using nodelet,  we will review and evaluate your repo. you should notice that we have  upgrade the SDK  to 1.5.4  for rplidar A1/A2.

bit-pirate, 2016-08-25:
Hi @kintzhao. This issue is just about adding a description for this repo and a link to a website, e.g. the ROS wiki. I have edited our fork as example:

[image]

Regarding the code changes in our repo, we will open separate PRs for the new features, such as the nodelet.

kintzhao, 2016-12-15:
I notice that your node about rplidar_ros  with nodelet has been deleted.

closed 2016-12-15 by kintzhao