robopeak/rplidar_ros #33: "min_ang" and "max_ang" parameters

davidbsp, 2017-01-18:
Similarly to the hokuyo_node (http://wiki.ros.org/hokuyo_node), it would be very useful to be able to change the minimum and maximum angles of the laserscans.

Since the robot has a 360º FOV, the need to filter out laser readings that correspond to objects that are behind the laser is a common issue for users.

Thanks!

kintzhao, 2017-02-17:
Thanks for your advice.    I fork and create two new branch   [feature/fixedAngleFilter ](https://github.com/kintzhao/rplidar_ros.git)
 [feature/fixedAngleFilter](https://github.com/kintzhao/rplidar_ros.git).  I will test it for better then merge  it, you can refer to my branch. Welcome to join us  to test and improve it.

closed 2017-07-18 by kintzhao

davidbsp, 2017-10-20:
Hi @kintzhao! Any reason for not merging this to the master branch?

subodh-malgonde, 2018-09-05:
@davidbsp Did you use the 2 branches to limit the angle of laser scans? If yes, can you let us know if it works?

davidbsp, 2018-09-05:
Hi @subodh-malgonde I did not. The issue I started was just a suggestion.
In my use case, I utilize the full 360º scan, and filter out the robot body at specific angle intervals, and then publish a filtered scan.