robopeak/rplidar_ros #34: 不能通过修改angle_compensate_offset来设置数据裁剪的起始位置

zuomaidi52, 2017-02-15:
在这里
`if ((angle_value - angle_compensate_offset) < 0)`
    `    angle_compensate_offset = angle_value;`
不管你修改成什么一个偏移值都会在第一次循环置零，所以我不能理解这里这句为什么要这样写，我将其改成了continue，则可以正确设置数据的偏移值了。

kintzhao, 2017-02-15:
[L269](https://github.com/robopeak/rplidar_ros/blob/4a54ec7a333c3ebb3bc968c72d19d2ae49c8b28b/src/node.cpp#L269 ) will set the param of angle_compensate_offset  by the first angle  grabbed nodes.

zuomaidi52, 2017-02-16:
现在我的需求是提取**lidar**的**90°～359°**的位置：
- 如果按照我修改后的代码（即换成**continue**）就可以通过设置**angle_compensate_offset**偏移地址为90°来获取正确的数据
- 如果我不修改的话，无论我设置**angle_compensate_offset**为多少，都不会对结果有变化。
- 那么，如果我不修改的话，应该要怎么改才能满足需求呢？

kintzhao, 2017-02-16:
angle_compensate_offset  not as the angle filter.  You can write  data filter when you process the data.
#33

zuomaidi52, 2017-02-17:
谢谢解答。

kintzhao, 2017-02-17:
I fork and create two new branch   [feature/fixedAngleFilter ](https://github.com/kintzhao/rplidar_ros.git)
 [feature/fixedAngleFilter](https://github.com/kintzhao/rplidar_ros.git).  I will test it for better then merge  it, you can refer to my branch. Welcome to join us  to test and improve it.

closed 2017-02-26 by kintzhao