robopeak/rplidar_ros #35: No scanning data received

sergiosierram, 2017-02-24:


kintzhao, 2017-02-26:
I ever works well at ubuntu 12.04 hydro with raspberry pi.  I can't  locate your problem through your error message , You can try to check the power of USB if it works at embedded computer .

sergiosierram, 2017-02-27:
Hi,

First, thanks for your time, answering my issue.
I was checking the power of the USB port, but it seems to be working fine, because the engine is running. Now, I am getting this when I run the launch file.

process[rplidarNode-1]: started with pid [3939]
RPLIDAR running on ROS package rplidar_ros
SDK Version: 1.5.7

And when I run this line: 
$ rostopic echo /scan

I don't get any data

Thanks.
Sergio Sierra

kintzhao, 2017-02-28:
Do you set the authority of usb:    sudo chmod a+rwx /dev/ttyUSB***

you can test it using linux-driver-test in the [SDK ](http://www.slamtec.com/cn/Lidar)at first.  such as simple_grabber (rplidar_sdk_v1.5.7\sdk\app\simple_grabber)

kintzhao, 2017-02-28:
Paste the message   :      ls -l /dev/ttyUSB*
the output message :    roslaunch rplidar_ros    rplidar.launch

sergiosierram, 2017-03-01:
I did set the authority of the usb port, using the udev rules provided in the package. It has all of the permissions and still not working.

kintzhao, 2017-03-02:
which version of rplidar (A1 / A2)?  I suggest that you support power to the motor of rpliadr by alone  in the embedded computer system.

kintzhao, 2017-03-13:
Do you solve your problem? I close it, I will reopen it if you still have any question.

closed 2017-03-13 by kintzhao