robopeak/rplidar_ros #37: No scan message received

Playfish, 2017-03-07:
Hi, I got a problem that rplidar A1 can used on PC but can not get scan data in Jetson Tk1.
Under is output message about rplidar a1.
ubuntu@tegra-ubuntu:~/catkin_ws/src/rplidar_ros$ roslaunch rplidar_ros view_rplidar.launch
... logging to /home/ubuntu/.ros/log/71dfd04e-1dd2-11b2-a063-f0421cf8ef80/roslaunch-tegra-ubuntu-4669.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:45894/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /rplidarNode/angle_compensate: True
 * /rplidarNode/frame_id: laser
 * /rplidarNode/inverted: False
 * /rplidarNode/serial_baudrate: 115200
 * /rplidarNode/serial_port: /dev/ttyUSB0

NODES
  /
    rplidarNode (rplidar_ros/rplidarNode)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[rplidarNode-1]: started with pid [4687]
process[rviz-2]: started with pid [4689]
RPLIDAR running on ROS package rplidar_ros
SDK Version: 1.5.7
RPLIDAR S/N: 98FEFBF2C8E49CCFC6E49FF1662C530D
Firmware Ver: 1.20
Hardware Rev: 0
RPLidar health status : 0
Everything is normal but cannot get data.
And also I give permissions for rplidar.
sudo chmod 777 /dev/ttyUSB0
Thanks!

kintzhao, 2017-03-07:
you can test it using linux-driver-test in the [SDK ](http://www.slamtec.com/cn/Lidar)at first.  such as simple_grabber (rplidar_sdk_v1.5.7\sdk\app\simple_grabber). 

I suggest that you support power to the motor of rpliadr by alone in the embedded computer system.

Playfish, 2017-03-07:
Hi! @kintzhao 
Thanks answer my issue, and I follow your suggestion and under is test output:
Output
==============
ubuntu@tegra-ubuntu:~/rplidar_sdk/sdk/output/Linux/Release$ ./simple_grabber /dev/ttyUSB1 115200 
RPLIDAR S/N: 98FEFBF2C8E49CCFC6E49FF1662C530D
Version: 1.5.7
Firmware Ver: 1.20
Hardware Rev: 0
RPLidar health status : OK. (errorcode: 0)
waiting for data...
***************************************************************************
***************************************************************************
***************************************************************************
***************************************************************************
***************************************************************************
***************************************************************************
***************************************************************************
***************************************************************************
***************************************************************************
***************************************************************************
***************************************************************************
***************************************************************************
***************************************************************************
***************************************************************************
***************************************************************************
***************************************************************************
***************************************************************************
***************************************************************************
***************************************************************************
***************************************************************************
---------------------------------------------------------------------------
Do you want to see all the data? (y/n) y
Error, cannot grab scan data.
=======================
Above is test result and when I use another rplidar A1 test output follow:
Another rplidar A1
======================
ubuntu@tegra-ubuntu:~/rplidar_sdk/sdk/output/Linux/Release$ ./simple_grabber /dev/ttyUSB1 115200
RPLIDAR S/N: 437CFBF2C8E49CCFC6E49FF1BDAE530D
Version: 1.5.7
Firmware Ver: 1.20
Hardware Rev: 0
RPLidar health status : OK. (errorcode: 0)
waiting for data...
                                                              *            
                                                             **            
                                                   *       *****           
                                                   *  *    *****           
                                                   *  *    ******          
         *  *                                      *  *    ******          
        ***** *                                    *  *    *******         
       ****** **                                   *  *    *********       
     ******** **                                   *  *    ***********     
************* **                                   *  *    ****************
************* **                                   *  *    ****************
************* ** *                                 ** *    ****************
************* ** *                                 ** *    ****************
**************** *                                 ** *    ****************
**************** *                                 ** *    ****************
**************** *                                 ** *  * ****************
**************** *                                 ** *  * ****************
**************** *                                 ** * ** ****************
**************** *                               * ** * ** ****************
***************************************************************************
---------------------------------------------------------------------------
Do you want to see all the data? (y/n) y
   theta: 0.08 Dist: 02721.00 
   theta: 1.00 Dist: 02723.00 
   theta: 1.94 Dist: 02725.00 
   theta: 2.86 Dist: 02725.00 
   theta: 3.80 Dist: 02725.00 
   theta: 4.72 Dist: 02732.00 
   theta: 5.64 Dist: 02732.00 
   theta: 6.58 Dist: 02737.00 
   theta: 7.50 Dist: 02752.00 
...
===============
Did Rplidar A1 using long time will reduce power of Rpliar A1?
Thanks!

Playfish, 2017-03-07:
Hi! @kintzhao 
I got the problem where cause I can not received scan data, Because I only have one USB3.0 so I used Hub to connect mobile base and rpliar, I check out cable individual and found its Hub problem, when I change another one its ok, Thanks and I will close this issue.

closed 2017-03-07 by Playfish

uffe, 2018-09-04:
I struggled with the same problem, and found my solution to be a dodgy USB cable.

XunshanMan, 2019-03-14:
I got the same problem and puzzled me for an afternoon. 
Finally I found the problem came from the bad connection of the line.

lenaNzrva, 2022-11-24:
Hi! 
I use rplidar a1m8
I have pretty same output using simple_grabber.exe

**Output:**
RPLIDAR S/N: 548DED93C0EA98C7C2E29EF5A6574060                                                                                                                                                                                                Version: 1.10.0                                                                                                                                                                                                                              Firmware Ver: 1.29                                                                                                                                                                                                                           Hardware Rev: 7                                                                                                                                                                                                                              RPLidar health status : OK. (errorcode: 0)                                                                                                                                                                                                   waiting for data...                                                                                                                                                                                                                          *WARN* YOU ARE USING DEPRECATED API: grabScanData(), PLEASE MOVE TO grabScanDataHq()                                                                                                                                                         *WARN* YOU ARE USING DEPRECATED API: ascendScanData(rplidar_response_measurement_node_t*, size_t), PLEASE MOVE TO ascendScanData(rplidar_response_measurement_node_hq_t*, size_t)                                                            ***************************************************************************                                                                                                                                                                  ***************************************************************************                                                                                                                                                                  ***************************************************************************                                                                                                                                                                  ***************************************************************************                                                                                                                                                                  ***************************************************************************                                                                                                                                                                  ***************************************************************************                                                                                                                                                                  ***************************************************************************                                                                                                                                                                  ***************************************************************************                                                                                                                                                                  ***************************************************************************                                                                                                                                                                  ***************************************************************************                                                                                                                                                                  ***************************************************************************                                                                                                                                                                  ***************************************************************************                                                                                                                                                                  ***************************************************************************                                                                                                                                                                  ***************************************************************************                                                                                                                                                                  ***************************************************************************                                                                                                                            [... cut ...]                                                                                                                                                                           Error, cannot grab scan data. 

If i use ultra_simple.exe or frame_grabber.exe I can't see any output. If I use ros I can see for distances only as inf.
I have checked my connection to the line even with multimeter everything seems okay.
A few days ago my liodar worked and I saw my data in the Rviz

Probably somebody can help me

tothaiduong, 2023-01-19:
Hi @lenaNzrva !
I use rplidar a1m8
I have the same problem as you when running simple_grabber.exe
If you fixed the error please help me fix it!

lenaNzrva, 2023-01-19:
Hi @tothaiduong !
My problem was the hardware one. The lidar just did not emit a laser beam. Write to your provider - my lidar was replased