robopeak/rplidar_ros #39: operation time out while processing "roslaunch rplidar_ros rolidar.launch"

didi1745, 2017-03-21:
when I input the "ls -l /dev |grep ttyUSB" command, I got "rplidar -> ttyUSB*", but when I am processing "roslaunch rplidar_ros rolidar.launch", I got some problems:

- process[rplidarNode-1]: started with pid [7979]
- RPLIDAR running on ROS package rplidar_ros
- SDK Version: 1.5.7
- Error, operation time out.
- [rplidarNode-1] process has died [pid 7979, exit code 255, cmd /home/ubuntu/turtlebot_ws/devel/lib/rplidar_ros/rplidarNode __name:=rplidarNode __log:=/home/ubuntu/.ros/log/b18a9430-0dd6-11e7-802c-00044b6351e1/rplidarNode-1.log].
- log file: /home/ubuntu/.ros/log/b18a9430-0dd6-11e7-802c-00044b6351e1/rplidarNode-1*.log
- all processes on machine have died, roslaunch will exit
- shutting down processing monitor...
- ... shutting down processing monitor complete
- done
-
BTW,I am using nvidia tx1, rplidar A1 :) 
Thank you!

kintzhao, 2017-03-21:
Does it always occur this problem?
you can test it using linux-driver-test in the [SDK](http://www.slamtec.com/cn/Lidar) at first. such as simple_grabber (rplidar_sdk_v1.5.7\sdk\app\simple_grabber)(not using ROS at first).
 
relevant issue :[issues/37](https://github.com/robopeak/rplidar_ros/issues/37)

didi1745, 2017-03-21:
- Ultra simple LIDAR data grabber for RPLIDAR.
- Version: 1.5.7
- Error, cannot get device info.

Above is my error info,what should i do now , Is the problem with rplidar and usb?

kintzhao, 2017-03-22:
firstly, check whether it is work well on PC with windows/linux  by formal SDK test. 
if it is ok, then  check the motor power on your TK1.  if not , contact with[ our support or sales. ](http://www.slamtec.com/cn/Home/About)

closed 2017-07-18 by kintzhao

3239611294, 2018-07-23:
Hello, have you solved your problem? I have the same problem.