robopeak/rplidar_ros #40: Setting motorPWM to low values (like 200) makes the node not publishing all points

rbiasini, 2017-03-23:
the slower the speed, the bigger the angle without points.

rbiasini, 2017-03-24:
Did some investigation today and found 3 bottle necks:

1- the function grabScanData accepts max node count as input. Currently is set to 360*2, which is too small for low rotational speeds.
`rplidar_response_measurement_node_t nodes[360*2];`

2- in rplidar_driver_serial.h there is a MAX_SCAN_NODES parameter set to 2048, which is also a bit low for very low rotational speeds (at motor PWM of 150 I see ~2400 nodes)
` MAX_SCAN_NODES = 2048*2`

3- in rplidar_driver_serial.h there is  _cached_scan_node_buf parameter initialized to size of 2048
`rplidar_response_measurement_node_t      _cached_scan_node_buf[2048*2];`

In conclusion, I'm not sure if this is an issue, as it's possible to change the parameters to see all points. However, it took me quite a while to figure out first that there was a problem and then to debug it. **A warning should at least be displayed when points are discarded.**

kintzhao, 2017-03-24:
which version of rplidar laser, rplidar A1 or A2.   which  range of frequency  do you  accept in your system?

rbiasini, 2017-03-24:
RPlidar A2M4 is the model. Not sure about the range of frequency: you mean which rotational frequency I'm ok with?

kintzhao, 2017-03-24:
@rbiasini  yeah,  make it( frequency or motor PWM value) be suitable for your system at first,  then consider your  investigation  and improve it in detail.

rbiasini, 2017-03-24:
@kintzhao , Let's say I want to set motor PWM to 200. Are the 3 bottle necks I mentioned above all you suggest to change to not drop points?

kintzhao, 2017-03-24:
@rbiasini It will not output  the data lower 3hz   by older firmware.  it will lower 2hz near 200 PWM.  we  suggest  that it is better at 10 hz .

kintzhao, 2017-03-24:
It is not suitable to change the code for lower frequency at official  node for your special demand. If it is needed, you can fork it and process it at your local repo.

closed 2017-07-18 by kintzhao