robopeak/rplidar_ros #44: Doubt about intensities array

jabrena, 2017-08-06:
Hi,

I am analyzing the `ROS` node for the `RPLIDAR` sensor and I would like to know what information store in the array about intensities:

https://github.com/robopeak/rplidar_ros/blob/master/src/node.cpp

```
scan_msg.intensities[node_count-1-i] = (float) (nodes[i].sync_quality >> 2);
```

What is the range of values that the sensor generate? In the SDK file: `rplidar_cmd.h` define the following structure:

``` 
typedef struct _rplidar_response_measurement_node_t {
  121:     _u8    sync_quality;      // syncbit:1;syncbit_inverse:1;quality:6;
```

I would like to know what possible values are populated to the ROS message: `sensor_msgs/LaserScan.h`

```
nodes[i].sync_quality >> 2
``` 

Many thanks in advance.

Juan Antonio

kintzhao, 2017-08-06:
It is a quality value of measure, same as intensities in the old SDK version..   But this value is invalid  value at current version of  SDK,  it is not well using it do anything at  current Version.

jabrena, 2017-08-06:
Hi @kintzhao,

Reading the sensor from a Java library, I can read the quality of the measure:

``` 
ev3dev#3|17:54:41.618 [Thread-1] INFO examples.Demo6 - Angle: 0, Distance: 0.0, Quality: 0
ev3dev#3|17:54:41.634 [Thread-1] INFO examples.Demo6 - Angle: 2, Distance: 0.0, Quality: 0
ev3dev#3|17:54:41.653 [Thread-1] INFO examples.Demo6 - Angle: 3, Distance: 0.0, Quality: 0
ev3dev#3|17:54:41.670 [Thread-1] INFO examples.Demo6 - Angle: 5, Distance: 0.0, Quality: 0
ev3dev#3|17:54:41.688 [Thread-1] INFO examples.Demo6 - Angle: 6, Distance: 0.0, Quality: 0
ev3dev#3|17:54:41.705 [Thread-1] INFO examples.Demo6 - Angle: 7, Distance: 0.0, Quality: 0
ev3dev#3|17:54:41.724 [Thread-1] INFO examples.Demo6 - Angle: 9, Distance: 0.0, Quality: 17
ev3dev#3|17:54:41.741 [Thread-1] INFO examples.Demo6 - Angle: 10, Distance: 0.0, Quality: 0
ev3dev#3|17:54:41.765 [Thread-1] INFO examples.Demo6 - Angle: 4, Distance: 283.575, Quality: 11
ev3dev#3|17:54:41.794 [Thread-1] INFO examples.Demo6 - Angle: 5, Distance: 282.4, Quality: 9
ev3dev#3|17:54:41.820 [Thread-1] INFO examples.Demo6 - Angle: 14, Distance: 0.0, Quality: 0
ev3dev#3|17:54:41.843 [Thread-1] INFO examples.Demo6 - Angle: 16, Distance: 0.0, Quality: 0
ev3dev#3|17:54:41.868 [Thread-1] INFO examples.Demo6 - Angle: 10, Distance: 290.5, Quality: 13
ev3dev#3|17:54:41.893 [Thread-1] INFO examples.Demo6 - Angle: 19, Distance: 0.0, Quality: 0
ev3dev#3|17:54:41.919 [Thread-1] INFO examples.Demo6 - Angle: 20, Distance: 0.0, Quality: 0
ev3dev#3|17:54:41.942 [Thread-1] INFO examples.Demo6 - Angle: 21, Distance: 0.0, Quality: 0
ev3dev#3|17:54:41.965 [Thread-1] INFO examples.Demo6 - Angle: 23, Distance: 0.0, Quality: 0
ev3dev#3|17:54:41.991 [Thread-1] INFO examples.Demo6 - Angle: 24, Distance: 0.0, Quality: 0
ev3dev#3|17:54:42.014 [Thread-1] INFO examples.Demo6 - Angle: 26, Distance: 0.0, Quality: 0
ev3dev#3|17:54:42.037 [Thread-1] INFO examples.Demo6 - Angle: 27, Distance: 0.0, Quality: 0
ev3dev#3|17:54:42.063 [Thread-1] INFO examples.Demo6 - Angle: 21, Distance: 136.675, Quality: 14
ev3dev#3|17:54:42.086 [Thread-1] INFO examples.Demo6 - Angle: 23, Distance: 133.125, Quality: 15
ev3dev#3|17:54:42.114 [Thread-1] INFO examples.Demo6 - Angle: 24, Distance: 130.05, Quality: 20
ev3dev#3|17:54:42.133 [Thread-1] INFO examples.Demo6 - Angle: 26, Distance: 129.725, Quality: 23
ev3dev#3|17:54:42.152 [Thread-1] INFO examples.Demo6 - Angle: 27, Distance: 129.4, Quality: 22
ev3dev#3|17:54:42.179 [Thread-1] INFO examples.Demo6 - Angle: 28, Distance: 131.425, Quality: 20
ev3dev#3|17:54:42.198 [Thread-1] INFO examples.Demo6 - Angle: 30, Distance: 133.225, Quality: 17
ev3dev#3|17:54:42.216 [Thread-1] INFO examples.Demo6 - Angle: 31, Distance: 138.675, Quality: 9
ev3dev#3|17:54:42.235 [Thread-1] INFO examples.Demo6 - Angle: 40, Distance: 0.0, Quality: 0
ev3dev#3|17:54:42.258 [Thread-1] INFO examples.Demo6 - Angle: 41, Distance: 0.0, Quality: 0
ev3dev#3|17:54:42.281 [Thread-1] INFO examples.Demo6 - Angle: 43, Distance: 0.0, Quality: 0
ev3dev#3|17:54:42.305 [Thread-1] INFO examples.Demo6 - Angle: 44, Distance: 0.0, Quality: 0
ev3dev#3|17:54:42.330 [Thread-1] INFO examples.Demo6 - Angle: 45, Distance: 0.0, Quality: 0
ev3dev#3|17:54:42.345 [Thread-1] INFO examples.Demo6 - Angle: 40, Distance: 131.7, Quality: 24
ev3dev#3|17:54:42.361 [Thread-1] INFO examples.Demo6 - Angle: 41, Distance: 133.4, Quality: 26
ev3dev#3|17:54:42.380 [Thread-1] INFO examples.Demo6 - Angle: 43, Distance: 135.9, Quality: 27
ev3dev#3|17:54:42.397 [Thread-1] INFO examples.Demo6 - Angle: 44, Distance: 139.075, Quality: 24
ev3dev#3|17:54:42.413 [Thread-1] INFO examples.Demo6 - Angle: 45, Distance: 138.55, Quality: 23
ev3dev#3|17:54:42.431 [Thread-1] INFO examples.Demo6 - Angle: 47, Distance: 137.025, Quality: 25
ev3dev#3|17:54:42.449 [Thread-1] INFO examples.Demo6 - Angle: 48, Distance: 141.025, Quality: 23
ev3dev#3|17:54:42.465 [Thread-1] INFO examples.Demo6 - Angle: 50, Distance: 147.45, Quality: 13
ev3dev#3|17:54:42.483 [Thread-1] INFO examples.Demo6 - Angle: 50, Distance: 459.425, Quality: 11
ev3dev#3|17:54:42.503 [Thread-1] INFO examples.Demo6 - Angle: 53, Distance: 134.25, Quality: 26
ev3dev#3|17:54:42.520 [Thread-1] INFO examples.Demo6 - Angle: 54, Distance: 131.95, Quality: 26
ev3dev#3|17:54:42.539 [Thread-1] INFO examples.Demo6 - Angle: 56, Distance: 121.0, Quality: 12
ev3dev#3|17:54:42.553 [Thread-1] INFO examples.Demo6 - Angle: 64, Distance: 0.0, Quality: 0
ev3dev#3|17:54:42.570 [Thread-1] INFO examples.Demo6 - Angle: 65, Distance: 0.0, Quality: 9
ev3dev#3|17:54:42.588 [Thread-1] INFO examples.Demo6 - Angle: 67, Distance: 0.0, Quality: 0
ev3dev#3|17:54:42.603 [Thread-1] INFO examples.Demo6 - Angle: 68, Distance: 0.0, Quality: 0
ev3dev#3|17:54:42.618 [Thread-1] INFO examples.Demo6 - Angle: 69, Distance: 0.0, Quality: 0
ev3dev#3|17:54:42.635 [Thread-1] INFO examples.Demo6 - Angle: 71, Distance: 0.0, Quality: 0
ev3dev#3|17:54:42.651 [Thread-1] INFO examples.Demo6 - Angle: 72, Distance: 0.0, Quality: 0
ev3dev#3|17:54:42.670 [Thread-1] INFO examples.Demo6 - Angle: 74, Distance: 0.0, Quality: 0
ev3dev#3|17:54:42.684 [Thread-1] INFO examples.Demo6 - Angle: 75, Distance: 0.0, Quality: 0
ev3dev#3|17:54:42.703 [Thread-1] INFO examples.Demo6 - Angle: 76, Distance: 0.0, Quality: 0
ev3dev#3|17:54:42.718 [Thread-1] INFO examples.Demo6 - Angle: 78, Distance: 0.0, Quality: 0
ev3dev#3|17:54:42.734 [Thread-1] INFO examples.Demo6 - Angle: 79, Distance: 0.0, Quality: 0
[... cut ...]
ev3dev#3|17:54:47.118 [Thread-1] INFO examples.Demo6 - Angle: 355, Distance: 0.0, Quality: 0
ev3dev#3|17:54:47.140 [Thread-1] INFO examples.Demo6 - Angle: 356, Distance: 0.0, Quality: 0
ev3dev#3|17:54:47.317 [Thread-1] INFO examples.Demo6 - Angle: 358, Distance: 0.0, Quality: 0
ev3dev#3|17:54:47.335 [Thread-1] INFO examples.Demo6 - Angle: 359, Distance: 0.0, Quality: 0
```  

I would like to know if I have to normalize the value to a determined range of values or, it is possible to send the value from the sensor directly.

Juan Antonio

kintzhao, 2017-09-25:
it is not meanful for real action.

jabrena, 2017-09-25:
So, in current slamtec implementation the intensities are not populated, isn't it?

I checked that hokuyo driver didn't populate the intensities array too:
https://github.com/rosjava/rosjava_extras/blob/kinetic/hokuyo/src/main/java/com/github/rosjava/rosjava_extras/hokuyo/LaserScanPublisher.java

Juan Antonio

kintzhao, 2017-09-25:
it is not meanful for the intensities array  at current version.

closed 2017-10-09 by kintzhao