robopeak/rplidar_ros #46: RPLidar A2 vibration

akatrevorjay, 2017-09-26:
Hi, it appears the head is not balanced in this unit? I just realized it makes my turtlebot vibrate so much that it screws up it's gyro which it uses in calculating /odom (it actually overwrites the encoder provided values in kobuki_driver). Is there a good solution to this? I've tried various methods of dampening to no avail, it's quite a bit of bad vibes coming off of this thing!

Looking forward to your reply.
Trevor.

akatrevorjay, 2017-10-01:
K, guess I'm going to have to return this unit, it's absolutely unusable in any environment that even remotely comes near a gyro. This is a crazy amount of inertia that it's exerting.

akatrevorjay, 2017-10-03:
Okay, I opened it up and I managed to find a nice place to place a weight in there!

akatrevorjay, 2017-10-03:
By the way guys, I think I see how this thing works now, eg how the head is powered as well as how it handles communication between the top and bottom, and I just have to say I'm really impressed by just how clever this design is. Well done.

Serveurperso, 2017-10-07:
I get the same problem because I need the maximum free run rotation speed (I get about >20Hz on mine).
I managed to add an small aluminium (2g) piece on the top with adhesive. now there is zero vibration (about 2 hours to find the right place and a laser to check vibration)

akatrevorjay, 2017-10-12:
Haha, yeah, I've been toying with a 1g. I'm glad you already found the
perfect weight, kudos!

Gonna find my weight set, will report back. For now I put my robot on a
Hokuyo until I get this fixed.

akatrevorjay, 2017-10-12:
Also, 20 hz? Not bad at all!
Really shows balancing it has quite the benefit.

Serveurperso, 2017-10-13:
Also you can fully test my application here :

https://www.serveurperso.com/?page=robot

Drag on the cam picture, and you get the 20Hz 2K sample/s lidar frame OSD.

It's a realtime web-robot working on One PIC32MX 80MHz Microcontroller doing a "very optimized SLAM-like" autonomous missions. with 8 cameras etc..

jomarlla, 2017-12-05:
Hi, Could you tell me where is the right position to put the 2g weigth in the top of RPLidar A2? Do you have any picture with the RPLidarA2 and the weight on it?
Thanks a lot
Jose

Serveurperso, 2017-12-05:
Hi Jose,

This is not a repeatable process, otherwise I think Slamtec would have included it in the production line. This is why you have to make the work yourself, individually on each RPLIDAR:)

Here is a picture of my two lidars after balancing the center of gravity for all speeds (I use them at 20Hz maximum speed) as you can see, it's totally different. however, the first lidar on the left is from an older batch, have they improved the manufacturing process?

[image]

Pascal

jomarlla, 2017-12-05:
Hi Pascal,

Your answer makes sense a lot
I understand now every unit is different so I will try to find the right position to balance it by my self.

Thanks a lot for your fast answer and pictures

Cheers,
Jose

kintzhao, 2018-04-17:
Thanks  for your comment for all user.

closed 2018-04-17 by kintzhao