robopeak/rplidar_ros #51: Incorrect valued of scan_duration

aalapshah12297, 2018-03-06:
At line no. 250 of src/node.cpp: 
`scan_duration = (end_scan_time - start_scan_time).toSec() * 1e-3;`
Why is the scan duration being multiplied by 0.001? ROS requires these times to be in seconds only (see the message [documentation](http://docs.ros.org/api/sensor_msgs/html/msg/LaserScan.html)). As a consequence, the `time_increment` field generated by the node is also incorrect.
If for some reason, this is deliberate, please add a comment to the code mentioning why (since it is not obvious). Otherwise, please fix this issue by changing that line to: 
`scan_duration = (end_scan_time - start_scan_time).toSec();`
I believe it could prevent problems foe people who are using this package as-is.

aalapshah12297, 2018-03-07:
Could you update this soon in the ROS package downloaded via apt-get too?

closed 2018-03-07 by aalapshah12297

kintzhao, 2018-03-08:
we will update it in the next big improve.