robopeak/rplidar_ros #53: what's the purpose of the angle compensate? And what's the principle of doing this?

linxigjs, 2018-03-19:
https://github.com/robopeak/rplidar_ros/blob/c67b213774397c600ddd0b48178e3fbe99c5c7c6/src/node.cpp#L258

kintzhao, 2018-03-19:
just for the principle of ros scan message  by fixed increased angle,  because the angle of scan in rplidar is not  fixed increased angle

linxigjs, 2018-03-19:
thx. But i still don't understand why rplidar is not fixed increased angle? And why this codeblock can realize the compensate and transform? Could you explain in detail?

kintzhao, 2018-03-19:
get angle and distance by grabScan();  
float angle = (float)((nodes[i].angle_q6_checkbit >> RPLIDAR_RESP_MEASUREMENT_ANGLE_SHIFT)/64.0f);

-----------
make float angle near integer index,  please read code

linxigjs, 2018-03-19:
Get it. THX.

closed 2018-03-19 by linxigjs