robopeak/rplidar_ros #55: rplidar don't work with cartographer

CodeIsWorld, 2018-04-07:
i try to use rplidar A1 in cartographer ,bu i got this
`rxh@ASUS-PC:~$ roslaunch cartographer_ros demo_revo_lds.launch
... logging to /home/rxh/.ros/log/d0dc176c-3a3c-11e8-9f3a-c8d3ffb49495/roslaunch-ASUS-PC-7602.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ASUS-PC:36319/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /use_sim_time: True

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[cartographer_node-1]: started with pid [7620]
WARNING: Package name "gpsCalibration" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
process[rviz-2]: started with pid [7633]
[ INFO] [1523090738.559865731]: I0407 16:45:38.000000  7620 configuration_file_resolver.cc:41] Found '/home/rxh/catkin_ws_cartographer/install_isolated/share/cartographer_ros/configuration_files/revo_lds.lua' for 'revo_lds.lua'.
[ INFO] [1523090738.560533819]: I0407 16:45:38.000000  7620 configuration_file_resolver.cc:41] Found '/home/rxh/catkin_ws_cartographer/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1523090738.560656306]: I0407 16:45:38.000000  7620 configuration_file_resolver.cc:41] Found '/home/rxh/catkin_ws_cartographer/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1523090738.560827372]: I0407 16:45:38.000000  7620 configuration_file_resolver.cc:41] Found '/home/rxh/catkin_ws_cartographer/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1523090738.560949598]: I0407 16:45:38.000000  7620 configuration_file_resolver.cc:41] Found '/home/rxh/catkin_ws_cartographer/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
F0407 16:45:38.561285  7620 lua_parameter_dictionary.cc:83] Check failed: status == 0 (2 vs. 0) [string "include "map_builder.lua"..."]:48: attempt to index global 'TRAJECTORY_BUILDER_2D' (a nil value)
[FATAL] [1523090738.561812234]: F0407 16:45:38.000000  7620 lua_parameter_dictionary.cc:83] Check failed: status == 0 (2 vs. 0) [string "include "map_builder.lua"..."]:48: attempt to index global 'TRAJECTORY_BUILDER_2D' (a nil value)
Got bus address:  "unix:abstract=/tmp/dbus-jNoDhQPKZ4,guid=fca53e1c6569377def3d342a5ac87f92" 
Connected to accessibility bus at:  "unix:abstract=/tmp/dbus-jNoDhQPKZ4,guid=fca53e1c6569377def3d342a5ac87f92" 
Registered DEC:  true 
*** Check failure stack trace: ***
    @     0x7f0e3cd89daa  (unknown)
    @     0x7f0e3cd89ce4  (unknown)
    @     0x7f0e3cd896e6  (unknown)
    @     0x7f0e3cd8c687  (unknown)
    @           0x677756  cartographer::common::(anonymous namespace)::CheckForLuaErrors()
    @           0x677cc6  cartographer::common::LuaParameterDictionary::LuaParameterDictionary()
    @           0x677e4d  cartographer::common::LuaParameterDictionary::LuaParameterDictionary()
    @           0x592682  cartographer_ros::LoadOptions()
    @           0x590313  cartographer_ros::(anonymous namespace)::Run()
    @           0x58d876  main
    @     0x7f0e3921af45  (unknown)
    @           0x590167  (unknown)
    @              (nil)  (unknown)
Registered event listener change listener:  true 
[cartographer_node-1] process has died [pid 7620, exit code -6, cmd /home/rxh/catkin_ws_cartographer/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/rxh/catkin_ws_cartographer/install_isolated/share/cartographer_ros/configuration_files -configuration_basename revo_lds.lua scan:=scan __name:=cartographer_node __log:=/home/rxh/.ros/log/d0dc176c-3a3c-11e8-9f3a-c8d3ffb49495/cartographer_node-1.log].
log file: /home/rxh/.ros/log/d0dc176c-3a3c-11e8-9f3a-c8d3ffb49495/cartographer_node-1*.log
`

kintzhao, 2018-05-31:
you can record it at  cartographer_ros , you can attach your launch file  for us to check it

LUOBO123LUOBO123, 2018-08-11:
How can you solve this problem?

closed 2018-08-31 by kintzhao