robopeak/rplidar_ros #57: rplidar a2 error operation time out

rplidarUser, 2018-05-09:
Hello,

I was trying to get a map with a rplidar a2 in combination with hector_mapping in Ubuntu, however when killing the rplidar launchfile something went wrong, the lidar was still rotating. After this I restarted the launchfile which eventually stopped the lidar, but now I am not able to restart the lidar again. When I try to start the lidar I receive this error message:

Error, operation time out.
[rplidarNode-2] process has died [pid 3114, exit code 255, cmd /home/pim/catkin_ws/devel/lib/rplidar_ros/rplidarNode __name:=rplidarNode __log:=/home/pim/.ros/log/c4f67012-5370-11e8-bb7d-9
cd21e0c8747/rplidarNode-2.log].
log file: /home/pim/.ros/log/c4f67012-5370-11e8-bb7d-9cd21e0c8747/rplidarNode-2*.log

I found out that it had something to do with the serial port but I cannot find a solution for this, could you help me?

kintzhao, 2018-05-31:
the port is still blocked , can you kill it again by id

kintzhao, 2018-05-31:
you can stop motor  by service when rplidar_ros run ok .   re-plugin the usb when kill the mode

closed 2018-05-31 by kintzhao