robopeak/rplidar_ros #62: Why does my lidar have no data back

KingAsher, 2018-06-02:
SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /rplidarNode/angle_compensate: True
 * /rplidarNode/frame_id: laser
 * /rplidarNode/inverted: False
 * /rplidarNode/serial_baudrate: 115200
 * /rplidarNode/serial_port: /dev/ttyUSB0

NODES
  /
    rplidarNode (rplidar_ros/rplidarNode)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://192.168.206.141:11311

process[rplidarNode-1]: started with pid [5049]
process[rviz-2]: started with pid [5050]
[ INFO] [1527909126.769626371]: RPLIDAR running on ROS package rplidar_ros. SDK Version:1.6.0
RPLIDAR S/N: 7633FBF2C8E49CCCC6E49FF188E1530E
[ INFO] [1527909129.278021707]: Firmware Ver: 1.20
[ INFO] [1527909129.278133668]: Hardware Rev: 0
[ INFO] [1527909129.280205710]: RPLidar health status : 0
[ INFO] [1527909129.807118185]: current scan mode: Express, max_distance: 16.0 m, Point number: 2.1K 

➜  ~ rostopic list
/initialpose
/move_base_simple/goal
/rosout
/rosout_agg
/scan
/tf
/tf_static
➜  ~ rosnode list
/rosout
/rplidarNode
/rviz

➜  ~ rostopic echo /scan         
header: 
  seq: 8952
  stamp: 
    secs: 1527910253
    nsecs: 981893616
  frame_id: "laser"
angle_min: -3.12413907051
angle_max: 3.14159274101
angle_increment: 3.39666002113e-19
time_increment: 6.64862573211e-21
scan_time: 0.122645497322
range_min: 0.15000000596
range_max: 16.0
ranges: []
intensities: []

KingAsher, 2018-06-02:
Rviz has no image display

kintzhao, 2018-06-04:
~ rostopic echo /scan                 //can not grabScan data , 
angle_increment: 3.39666002113e-19

is it ok on windows SDK ?

erwincoumans, 2018-06-07:
With latest github revision my rpLidar A1 doesn't get any data back either. 
Checking out an older revision makes it work:
git checkout c67b213774397c600ddd0b48178e3fbe99c5c7c6
catkin_make

Also, why can't we update firmware to 1.24 for rpLidar A1?

kintzhao, 2018-06-08:
@erwincoumans   [SDK and Firmware](http://www.slamtec.com/cn/Support#rplidar-a1 )    contact with support@slamtec.com if  can not update firmware

KingAsher, 2018-06-08:
@kintzhao  frame_grabber can grab Scan data on windows. 
my rpLidar A1 serial number : 7633FBF2C8E49CCCC6E49FF188E1530E 
Cannot support updates.

KingAsher, 2018-06-08:
➜  launch git:(master) ✗ # roslaunch test_rplidar.launch
... logging to /home/lee/.ros/log/c61543da-6b01-11e8-87d9-000c29a75703/roslaunch-virtual-3190.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.206.141:43605/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /rplidarNode/angle_compensate: True
 * /rplidarNode/frame_id: laser
 * /rplidarNode/inverted: False
 * /rplidarNode/serial_baudrate: 115200
 * /rplidarNode/serial_port: /dev/ttyUSB0

NODES
  /
    rplidarNode (rplidar_ros/rplidarNode)
    rplidarNodeClient (rplidar_ros/rplidarNodeClient)

auto-starting new master
process[master]: started with pid [3200]
ROS_MASTER_URI=http://192.168.206.141:11311

setting /run_id to c61543da-6b01-11e8-87d9-000c29a75703
process[rosout-1]: started with pid [3213]
started core service [/rosout]
process[rplidarNode-2]: started with pid [3229]
process[rplidarNodeClient-3]: started with pid [3238]
[ INFO] [1528451600.832108377]: RPLIDAR running on ROS package rplidar_ros. SDK Version:1.6.0
RPLIDAR S/N: 7633FBF2C8E49CCCC6E49FF188E1530E
[ INFO] [1528451603.344005222]: Firmware Ver: 1.20
[ INFO] [1528451603.344070368]: Hardware Rev: 0
[ INFO] [1528451603.346103099]: RPLidar health status : 0
[ INFO] [1528451603.873247656]: current scan mode: Express, max_distance: 16.0 m, Point number: 2.1K 
[ INFO] [1528451604.640909309]: I heard a laser scan laser[-2147483648]:
[ INFO] [1528451604.641007695]: angle_range, -178.999983, 180.000005
[ INFO] [1528451604.778994457]: I heard a laser scan laser[-2147483648]:
[ INFO] [1528451604.779042288]: angle_range, -178.999983, 180.000005
[ INFO] [1528451604.901531902]: I heard a laser scan laser[-2147483648]:
[ INFO] [1528451604.901610325]: angle_range, -178.999983, 180.000005
[ INFO] [1528451605.026015024]: I heard a laser scan laser[-2147483648]:
[ INFO] [1528451605.026257722]: angle_range, -178.999983, 180.000005
[ INFO] [1528451605.146814987]: I heard a laser scan laser[-2147483648]:
[ INFO] [1528451605.146915161]: angle_range, -178.999983, 180.000005
[ INFO] [1528451605.284902965]: I heard a laser scan laser[-2147483648]:
[ INFO] [1528451605.285000523]: angle_range, -178.999983, 180.000005
[ INFO] [1528451605.409098450]: I heard a laser scan laser[-2147483648]:
[ INFO] [1528451605.409206494]: angle_range, -178.999983, 180.000005
[ INFO] [1528451605.530219223]: I heard a laser scan laser[-2147483648]:
[ INFO] [1528451605.530296804]: angle_range, -178.999983, 180.000005
[ INFO] [1528451605.653018856]: I heard a laser scan laser[-2147483648]:
[ INFO] [1528451605.653066978]: angle_range, -178.999983, 180.000005
[ INFO] [1528451605.775758016]: I heard a laser scan laser[-2147483648]:
[ INFO] [1528451605.775822312]: angle_range, -178.999983, 180.000005
[ INFO] [1528451605.913705917]: I heard a laser scan laser[-2147483648]:
[ INFO] [1528451605.913754163]: angle_range, -178.999983, 180.000005
[ INFO] [1528451606.036383126]: I heard a laser scan laser[-2147483648]:
[ INFO] [1528451606.036459420]: angle_range, -178.999983, 180.000005
[ INFO] [1528451606.159034894]: I heard a laser scan laser[-2147483648]:
[ INFO] [1528451606.159112535]: angle_range, -178.999983, 180.000005
[ INFO] [1528451606.283319492]: I heard a laser scan laser[-2147483648]:
[ INFO] [1528451606.283432295]: angle_range, -178.999983, 180.000005
[ INFO] [1528451606.419692330]: I heard a laser scan laser[-2147483648]:
[ INFO] [1528451606.419833484]: angle_range, -178.999983, 180.000005
[ INFO] [1528451606.542278077]: I heard a laser scan laser[-2147483648]:
[ INFO] [1528451606.542327328]: angle_range, -178.999983, 180.000005
[ INFO] [1528451606.665069367]: I heard a laser scan laser[-2147483648]:
[ INFO] [1528451606.665117786]: angle_range, -178.999983, 180.000005
^C[ INFO] [1528451606.787633009]: I heard a laser scan laser[-2147483648]:
[ INFO] [1528451606.787726151]: angle_range, -178.999983, 180.000005
[rplidarNodeClient-3] killing on exit
[rplidarNode-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

kintzhao, 2018-06-26:
thanks, fixed https://github.com/robopeak/rplidar_ros/commit/f12589d98dba50ce464318c20ee007c7d9ffb3ad

closed 2018-08-31 by kintzhao