robopeak/rplidar_ros #66: crash in ubuntu16.04 ROS kinetic

gongyue666, 2018-08-16:
hi, i have an a3 rplidar, when roslaunch rplidar_ros view_rplidar_a3.launch, node crash.log as below:
  RPLIDAR S/N: E5BB9AF2C1EA9FC3A2EB92F176683C01
[ INFO] [1534383529.782389325]: Firmware Ver: 1.24
[ INFO] [1534383529.782420782]: Hardware Rev: 6
[ INFO] [1534383529.783272236]: RPLidar health status : 0
[ INFO] [1534383530.397773691]: current scan mode: Sensitivity, max_distance: 25.0 m, Point number: 15.9K , angle_compensate: 4
[rplidarNode-2] process has died [pid 1932, exit code -7, cmd /home/gy/navigation/devel/lib/rplidar_ros/rplidarNode __name:=rplidarNode __log:=/home/gy/.ros/log/1f226ca4-a0f5-11e8-bbf6-f0421c857a46/rplidarNode-2.log].
log file: /home/gy/.ros/log/1f226ca4-a0f5-11e8-bbf6-f0421c857a46/rplidarNode-2*.log

Piroro-hs, 2018-08-27:
Hi, I faced the same issue sometimes, and using `catkin_make_isolated` instead of `catkin_make` always fixed the issue.

gongyue666, 2018-08-27:
@Piroro-hs It's worked for me too, thx~

kintzhao, 2018-08-31:
@gongyue666  is it ok now?  which platform ,PC or arm-board？

Vuwij, 2018-10-04:
Hello, I have a similar bug, here are the steps I ran

```
sudo apt-get update
sudo apt-get install ros-kinetic-rplidar-ros
roslaunch rplidar_ros rplidar_a3.launch
```
Here is the results:
```
[ INFO] [1538621877.166500709]: RPLIDAR running on ROS package rplidar_ros. SDK Version:1.7.0
RPLIDAR S/N: 96C59AF2C1EA9FC0A2EB92F105163C02
[ INFO] [1538621877.671937261]: Firmware Ver: 1.24
[ INFO] [1538621877.672055649]: Hardware Rev: 6
[ INFO] [1538621877.673490754]: RPLidar health status : 0
[ INFO] [1538621878.303578494]: current scan mode: Sensitivity, max_distance: 25.0 m, Point number: 15.9K , angle_compensate: 4
[rplidarNode-1] process has died [pid 22288, exit code -7, cmd /opt/ros/kinetic/lib/rplidar_ros/rplidarNode __name:=rplidarNode __log:=/home/rr/.ros/log/40411be2-df7a-11d4-8baf-b827eb187a25/rplidarNode-1.log].
log file: /home/rr/.ros/log/40411be2-df7a-11d4-8baf-b827eb187a25/rplidarNode-1*.log
```
I've tried compiling from source, and it works, this indicates that the Debian packages are not updated. My system is Intel NUC running Ubuntu 16.04

Thank you

neomanic, 2018-10-16:
I'm getting the same crash, both with the apt-installed package, and also with my compiled one in a vendor catkin workspace.

I finally solved it by building using `catkin_make_isolated`, many thanks @Piroro-hs.

```
$ apt show ros-kinetic-rplidar-ros
Package: ros-kinetic-rplidar-ros
Version: 1.7.0-0xenial-20180824-112159-0800
```

and then when running:
```
$ roslaunch rplidar_ros rplidar_a3.launch
[ ... ]
process[rplidar-1]: started with pid [30903]
[ INFO] [1539729453.736844593]: RPLIDAR running on ROS package rplidar_ros. SDK Version:1.7.0
RPLIDAR S/N: E6FC9AF2C1EA9FC3A2EB92F1752F3C01
[ INFO] [1539729454.241341792]: Firmware Ver: 1.24
[ INFO] [1539729454.241410282]: Hardware Rev: 6
[ INFO] [1539729454.242437629]: RPLidar health status : 0
[ INFO] [1539729454.779141734]: current scan mode: Sensitivity, max_distance: 25.0 m, Point number: 15.9K , angle_compensate: 4
[rplidar-1] process has died [pid 30903, exit code -7, cmd /opt/ros/kinetic/lib/rplidar_ros/rplidarNode __name:=rplidar __log:=/home/sorter/.ros/log/1986cd64-d0fa-11e8-9fcb-0fb7c8ec8546/rplidar-1.log].
log file: /home/sorter/.ros/log/1986cd64-d0fa-11e8-9fcb-0fb7c8ec8546/rplidar-1*.log
```

kintzhao, 2019-01-15:
https://github.com/robopeak/rplidar_ros/pull/82

Zacrain, 2019-04-03:
Thank you @neomanic ! Fixed it for me. Installing the package via apt led to the same problem. Compiling it from source using `catkin_make_isolated` fixed it.

I'm running ROS Kinetic under Ubuntu 16.04.

pjimenezmateo, 2019-12-16:
> #82
> [param]add param angle_compensate_multiple
> Default angle_compensate_multiple(Running at 10HZ): 16K -- 4 8K -- 2 4K -- 1
> angle_compensate_multiple is min(setting value, default value)

Thank you very much! Using your patch finally makes it work on Ubuntu 18.04 with kinetic!