robopeak/rplidar_ros #67: Crash on Raspberry Pi

Piroro-hs, 2018-08-17:
Hello.

I cloned this repo, and `catkin_make`d on both my PC and Raspberry Pi. The `rplidarNode` is running well on my Ubuntu 16.04 x64 PC, but always crash on Raspberry Pi. Below is a error message.

```
process[rplidarNode-1]: started with pid [1566]
[ INFO] [1534499013.847457964]: RPLIDAR running on ROS package rplidar_ros. SDK Version:1.7.0
RPLIDAR S/N: 97DD9AF2C1EA9FC0A2EB92F1040E3C02
[ INFO] [1534499014.366249529]: Firmware Ver: 1.24
[ INFO] [1534499014.366457706]: Hardware Rev: 6
[ INFO] [1534499014.368799211]: RPLidar health status : 0
[ INFO] [1534499015.067182568]: current scan mode: Sensitivity, max_distance: 25.0 m, Point number: 15.9K , angle_compensate: 4
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
[rplidarNode-1] process has died [pid 1566, exit code -6, cmd /home/pi/catkin_ws/devel/lib/rplidar_ros/rplidarNode __name:=rplidarNode __log:=/home/pi/.ros/log/6f73eaa0-a1c9-11e8-8585-7085c2230810/rplidarNode-1.log].
log file: /home/pi/.ros/log/6f73eaa0-a1c9-11e8-8585-7085c2230810/rplidarNode-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Also it sometimes throws `std::length_error`, instead of `std::bad_alloc`.

Thank you.

Piroro-hs, 2018-08-17:
Related? https://github.com/Slamtec/rplidar_ros/issues/1

kintzhao, 2018-08-30:
https://github.com/robopeak/rplidar_ros/blob/master/src/node.cpp#L270   you can try to change the code by an const angle_compensate_multiple ;
//angle_compensate_multiple = (int)(1000*1000/current_scan_mode.us_per_sample/10.0/360.0);
angle_compensate_multiple = 1;

closed 2019-01-15 by kintzhao