robopeak/rplidar_ros #69: [ERROR] [1535647334.151867876]: Error, cannot bind to the specified serial port /dev/ttyUSB0.

burakekincioglu, 2018-08-30:


burakekincioglu, 2018-08-30:
How to remap the USB serial port name
Maybe you should change the usb device port mode with the authority of read and write, more than make the port remap to an fixed name. install USB port remap: ./scripts/create_udev_rules.sh

chaeck the remap use command: ls -l /dev | grep ttyUSB usb_port_remap

Once you have change the USB port remap, you can change the launch file about the serial_port value.
<param name="serial_port" type="string" value="/dev/rplidar"/>

I SOLVED MY PROBLEM WİTH COMANDS THAT ABOVE.

kintzhao, 2018-08-31:
it has the authority 0777  ./scripts/create_udev_rules.sh

KERNEL=="ttyUSB*", ATTRS{idVendor}=="10c4", ATTRS{idProduct}=="ea60", MODE:="0777", SYMLINK+="rplidar"

closed 2018-08-31 by kintzhao