robopeak/rplidar_ros #77: What is the correct orientation of RPLidar?

subodh-malgonde, 2018-10-24:
I want to install the sensor on my robot such that there is no transform from base_link to laser i.e. no translation and no rotation. Should the tapering end be facing the back of my robot, as shown in the picture?

[image]

Or should I place it the other way round?

GYKrauss, 2019-01-09:
The green arrow marked "x" is facing downwards. This means that the motor end is really the "front." If you want to avoid any tf, it must be placed "backwards."

kintzhao, 2019-01-15:
keep your any install by configure your TF-tree frame:   base_link -> laser

such as: [(13/57)](http://www.robotics.sei.ecnu.edu.cn/ros2016/docs/RPLidar%20For%20ROS%20Based%20SLAM%20and%20Navigation_slamtec_KintZhao.pdf)

closed 2019-01-15 by kintzhao

alrad32, 2023-09-14:
Hello, 
Where should the origin of the RPLidar be in the Z-axis? Is it right on the same plane as the laser transmitter and receiver?