robopeak/rplidar_ros #78: RESULT_OPERATION_TIMEOUT on raspberry pi 3 B

ameykasar, 2018-10-29:
I am trying to interface **RPLIDAR A1** with my **Raspberry Pi 3 B 2015** Model running ubuntu 16.04 and ROS kinetic.
I am receiving the following error: **_RESULT_OPERATION_TIMEOUT_** when I use _roslaunch rplidar_ros rplidar.launch_
I tried using multiple OS images and the standalone SDK but the error persists.
It works fine on my PC running Ubuntu 16.04 and ROS kinetic with same USB cable.
Is this due to Raspberry Pi 3 B only having USB 2.0 ports? Help me find a solution

ameykasar, 2018-10-29:
@kintzhao

ArthurGuy, 2018-11-14:
I am experiencing the same problem with the same setup.

kintzhao, 2019-01-15:
https://github.com/Slamtec/rplidar_ros/issues/1

check your USB-line's power  or your raspberry's power

ArthurGuy, 2019-01-15:
For me it turned out to be a faulty unit, as soon as it was replaced everything worked as expected.

closed 2019-01-16 by kintzhao

ameykasar, 2019-02-01:
> For me it turned out to be a faulty unit, as soon as it was replaced everything worked as expected.

@ArthurGuy  Can you elaborate? By unit do you mean Raspberry Pi or RPLIDAR?