robopeak/rplidar_ros #79: A3  crash with scan_mode Sensitivity

winf, 2018-11-01:
[image]
However ，scan_mode with satability is working.

winf, 2018-11-01:
disable angle_compensate could resolve this

winf, 2018-11-01:
maybe “ __attribute__((packed)) ” in rplidar_cmd.h #line204  cause this behave.
https://stackoverflow.com/questions/8568432/is-gccs-attribute-packed-pragma-pack-unsafe

ramonvp, 2019-02-11:
Hi! Any news on this? I am also experiencing the same crash. It does not happen if you modify the launch file and use "Standard" as scan_mode instead of "Sensitivity"

cjmahieu, 2019-02-27:
I found that both the Standard and Express mode work. However I would like the Sensitivity mode to work as well..

Disabling angle compensate prevents the crash from happening but makes the scan data unusable as shown below:

[image]

(With express / standard option it looks as follows:)

[image]

ameykasar, 2019-02-27:
Had the same problem. Editing the scan mode in launch file (as mentioned above) helped.
The problem went away completely when I switched to another USB cable. Digging more I found out the cable was not capable of providing enough stable power for the sensor.
Observe the laser does not light up when you get the error as the power is not sufficient.

TLDR: CHANGE USB CABLE/ CHECK POWER DELIVERED TO THE SENSOR

deb0ch, 2019-03-25:
Looks a lot like https://github.com/Slamtec/rplidar_ros/issues/1 and fixed by https://github.com/Slamtec/rplidar_ros/pull/10 (although I'm not comfortable with the arbitrary number chosen to extend the buffer).

I suspect this repo is deprecated now that the Slamtec one is there so this is FYI