robopeak/rplidar_ros #91: catkin_make fails to build, freshly installed ros

Jiwii, 2019-05-01:
After i cloned the rplidar to the src foulder in my catkin_ws i run the command catkin_make in the ws and i get this error. I tried redoing all the steps and got the same results. I used ( http://wiki.ros.org/Installation/Windows ) to install ros and ( http://wiki.ros.org/ROS/Tutorials/CreatingPackage ) to make a ws. 
I have checked my ROS_PACKAGE_PATH: 
`ROS_PACKAGE_PATH=C:\opt\ros\melodic\x64\share` 

```
Base path: C:\Users\Jan-Willem\catkin_tutorial
Source space: C:\Users\Jan-Willem\catkin_tutorial\src
Build space: C:\Users\Jan-Willem\catkin_tutorial\build
Devel space: C:\Users\Jan-Willem\catkin_tutorial\devel
Install space: C:\Users\Jan-Willem\catkin_tutorial\install
####
#### Running command: "nmake cmake_check_build_system" in "C:\Users\Jan-Willem\catkin_tutorial\build"
####

Microsoft (R) Program Maintenance Utility Version 14.16.27030.1
Copyright (C) Microsoft Corporation.  All rights reserved.

####
#### Running command: "nmake" in "C:\Users\Jan-Willem\catkin_tutorial\build"
####

Microsoft (R) Program Maintenance Utility Version 14.16.27030.1
Copyright (C) Microsoft Corporation.  All rights reserved.

[ 18%] Built target _setup_util.py_executable_install_python
[ 27%] Building CXX object rplidar_ros/CMakeFiles/rplidarNode.dir/src/node.cpp.obj
node.cpp
c:\users\jan-willem\catkin_tutorial\src\rplidar_ros\sdk\include\rplidar_protocol.h(59): warning C4200: nonstandard extension used: zero-sized array in struct/union
c:\users\jan-willem\catkin_tutorial\src\rplidar_ros\sdk\include\rplidar_protocol.h(59): note: This member will be ignored by a defaulted constructor or copy/move assignment operator
c:\users\jan-willem\catkin_tutorial\src\rplidar_ros\sdk\include\rplidar_cmd.h(256): warning C4200: nonstandard extension used: zero-sized array in struct/union
c:\users\jan-willem\catkin_tutorial\src\rplidar_ros\sdk\include\rplidar_cmd.h(256): note: This member will be ignored by a defaulted constructor or copy/move assignment operator
C:\Users\Jan-Willem\catkin_tutorial\src\rplidar_ros\src\node.cpp(78): warning C4305: '=': truncation from 'double' to 'sensor_msgs::LaserScan_<std::allocator<void>>::_range_min_type'
C:\Users\Jan-Willem\catkin_tutorial\src\rplidar_ros\src\node.cpp(321): warning C4305: 'initializing': truncation from 'double' to 'float'
C:\Users\Jan-Willem\catkin_tutorial\src\rplidar_ros\src\node.cpp(327): error C2131: expression did not evaluate to a constant
C:\Users\Jan-Willem\catkin_tutorial\src\rplidar_ros\src\node.cpp(325): note: failure was caused by a read of a variable outside its lifetime
C:\Users\Jan-Willem\catkin_tutorial\src\rplidar_ros\src\node.cpp(325): note: see usage of 'angle_compensate_multiple'
C:\Users\Jan-Willem\catkin_tutorial\src\rplidar_ros\src\node.cpp(337): error C3863: array type 'rplidar_response_measurement_node_hq_t [angle_compensate_nodes_count]' is not assignable
C:\Users\Jan-Willem\catkin_tutorial\src\rplidar_ros\src\node.cpp(352): warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data
C:\Users\Jan-Willem\catkin_tutorial\src\rplidar_ros\src\node.cpp(367): warning C4305: 'initializing': truncation from 'double' to 'float'
NMAKE : fatal error U1077: 'C:\PROGRA~2\MIB055~1\2017\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\Hostx64\x64\cl.exe' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
Invoking "nmake" failed
```

asbird, 2019-07-28:
Hi, did you find solution to your problem already ?

mhendersonccr, 2019-09-26:
i get the exact same error. not sure what im doing wrong

ishanijanveja, 2020-03-06:
Any updates on this? Seems like this ROS package wasn't tested on windows.