robopeak/rplidar_ros #92: Error: Cannot launch node of type [rplidar_ros/rplidarNode];can't locate node[rplidarNode]

AndreaGonsalves, 2019-05-25:
Hey
I have tried the following commands
cd catkin_ws
source devel/setup.bash
roslaunch rplidar_ros rplidar.launch

I get the following results:

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /rplidarNode/angle_compensate: True
 * /rplidarNode/frame_id: laser
 * /rplidarNode/inverted: False
 * /rplidarNode/serial_baudrate: 115200
 * /rplidarNode/serial_port: /dev/ttyUSB0

NODES
  /
    rplidarNode (rplidar_ros/rplidarNode)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [4869]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 9e5af918-7ea9-11e9-bd54-7cdd90efc4a0
process[rosout-1]: started with pid [4880]
started core service [/rosout]
**ERROR: cannot launch node of type [rplidar_ros/rplidarNode]: can't locate node [rplidarNode] in package [rplidar_ros]**
process[rviz-3]: started with pid [4887]
libEGL warning: DRI2: failed to authenticate

I have uninstalled and reinstalled the package. Any help would be great!

chriszhuuu, 2019-08-06:
I got the same issue

chriszhuuu, 2019-08-06:
is anyone can help

yuzhenning, 2019-11-28:
hello! I also meet the same problem.
 It seems like fixed in following way.
Please locate your ROS workspace firstly.
In my robot, the ros workspace is `/turtlebot_ws`
re-init your workspace as following cammand :
```
$ cd ~/turtlebot_ws/src
$ catkin_init_workspace
```

```
$ cd ~/turtlebot_ws/
$ catkin_make
```
Then, please check rplidar performance by following command:
`$ roslaunch rplidar_ros view_rplidar.launch`
In my robot , the problem fixed.
I hope it will help you. 
Have a nice day!

yuzhenning, 2019-11-28:
> is anyone can help
hello! I also meet the same problem.
It seems like fixed in following way.
Please locate your ROS workspace firstly.
In my robot, the ros workspace is /turtlebot_ws
re-init your workspace as following cammand :

$ cd ~/turtlebot_ws/src
$ catkin_init_workspace

$ cd ~/turtlebot_ws/
$ catkin_make

Then, please check rplidar performance by following command:
$ roslaunch rplidar_ros view_rplidar.launch
In my robot , the problem fixed.
I hope it will help you.
Have a nice day!

aksharc2, 2021-07-29:
Hi, I tried this method but it did not work.
When I do catkin_make the node.cpp and client.cpp are not made executable....
Any suggestions how to resolve this..