robopeak/rplidar_ros #95: Can not start scan: 80008000!

Gowresh7, 2019-07-04:
I am using RPLiDAR A1 in Raspberry Pi 3B with ROS. I am getting this error when I launch the rplidar node with my launch file. How to rectify this error?

garyrightbot, 2020-10-14:
In my case this was due to faulty cables.

mdthabrez, 2023-10-29:
@Gowresh7 i am also facing the same problem , did u solve yours .
Please give a solution

Gowresh-MARC, 2023-10-30:
So I identified the problem was primarily due to the inadequate power from the usb ports. Once I switched to a external powered usb hub, it was solved.

> @Gowresh7 i am also facing the same problem , did u solve yours . Please give a solution

mdthabrez, 2023-10-30:
i am making a raspberry pi + rplidar a1 based autonomous robot car. I am supplying a 5.15V to the raspberry pi through a 12V lipo battery but it still not working . How much volt would you recommend for the robot car using raspberry pi 3 b + rplidar a1m8 + arduino (as motor controlller) . both arduino and rplidar are powered by through the raspberry pi .