robopeak/rplidar_ros #101: and of course it does not compile

phil123456, 2020-03-05:
Hi

I just try catkin_make and it fails to compile (I use ROS melodic on a RPI4)

any help appreciated

```
-- +++ processing catkin package: 'rplidar_ros'
-- ==> add_subdirectory(rplidar_ros)
-- Could NOT find sensor_msgs (missing: sensor_msgs_DIR)
-- Could not find the required component 'sensor_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /home/pi/ros_catkin_ws/devel/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "sensor_msgs" with
  any of the following names:

    sensor_msgsConfig.cmake
    sensor_msgs-config.cmake

  Add the installation prefix of "sensor_msgs" to CMAKE_PREFIX_PATH or set
  "sensor_msgs_DIR" to a directory containing one of the above files.  If
  "sensor_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  rplidar_ros/CMakeLists.txt:12 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/ros_catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/ros_catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [Makefile:7474: cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
```

thanks

phil123456, 2020-03-12:
after a lot of digging and help from "Ros Answers"...

- I cloned manualy some packages
- I ended up using ros build tools
- I added a dependecy to std_srv package

Smitty-44, 2020-05-14:
Hi Phil,
I have been having problems with the catkin_make of rplidar_ros, too.
I have ROS Melodic, RosOnWindows.  Been working fine for ~6 months.
I actually have the exact same problem as issue #91 .
No one has responded to that issue.
But, I saw that you had made it through your problem.
I read your conversation concerning adding the dependency for std_srvs, etc. (ROS answers)
So, I already modified my code to do that.
I was wondering what packages did you clone manually?
And how did you get past the node.cpp errors?
Here is the link to 91: https://github.com/robopeak/rplidar_ros/issues/91

Any help would really be greatly appreciated.  Thank you, Smitty

DariusAvalon, 2020-11-14:
Having this same issue under ROS Noetic on Ubuntu 20.04. Would appreciate a response from the maintainers.

mshue, 2021-02-07:
I solved this problem on my RPi4 by doing "git clone https://github.com/ros/common_msgs.git" into my ~/ros_catkin_ws/src/ros directory. Then I rebuilt everything by running catkin_make from ~/ros_catkin_ws