robopeak/rplidar_ros #103: no ttyUSB,cannot communicate with lidar

Shelfcol, 2020-07-23:
I plugin the lidar in my ubuntu16.04, but no ttyUSB file exists in /dev  directory, so I cannot bind ttyUSB as well. And I also execute  /scripts/create_udev_rules.sh,but dont help. Do somebody know solutions?  thanks.

bramburn, 2020-07-23:
Make sure your usb is outputting 5v some machines like the old Mac need an external source for it to pick it up.

Shelfcol, 2020-07-23:
Same computer with two system(windows ans ubuntu)，And the lidar worked on  windows,but failed on ubuntu

Shelfcol, 2020-07-23:
I suddenly found it 's the cable problem. I changed a cable, But I found long cable was not OK, said,  can not start scan:80008000. so I used a short one, it worked. Sorry to interrupt.

bramburn, 2020-07-27:
hmm interesting. you should close the ticket if you can.